adafruit/SdFat #10: Solution to support Uno WiFi R2

Travelbacon, 2021-12-28:
**Is your feature request related to a problem? Please describe.**
To make the Arduin Uno WiFi R2 work with this library.

**Describe the solution you'd like**
To change the following lines in SdSpiDriver.h 
```
//==============================================================================
// Use of in-line for AVR to save flash.
#ifdef __AVR__
//------------------------------------------------------------------------------
```

with

```
//==============================================================================
// Use of in-line for AVR to save flash.
#ifdef __AVR__
#if defined (__AVR_ATmega4809__)
#define SPDR SPI0_DATA
#define SPSR SPI0_INTFLAGS
#define SPIF 7
#endif // __AVR_ATmega4809__
//------------------------------------------------------------------------------
```

This solution was first posted by thorx on your [forum](https://forums.adafruit.com/viewtopic.php?f=47&t=161677#p799163)  It works, and I see that other people did this to make the SD card reader work with the Uno WiFi R2. It also enables to make the 2.8" TFT work with the Uno WiFi R2, where more people are struggling with.

I hope that this feature will be accepted and if a pull request is desired.
Let me know if this feature

ladyada, 2021-12-28:
ohhh nifty, please do a PR :) thank you!!!

Travelbacon, 2022-01-03:
Is merged in #11

closed 2022-01-03 by Travelbacon