adafruit/SdFat #15: slow speeds on stm32, and bug when buffer is bigger than file

nablabla, 2022-12-29:
Hi, I like to read stuff fast, but I think something is not right here.
STM32f103 (bluepill)
Arduino framework, PlatformIO 
lib_deps = 
	adafruit/SdFat - Adafruit Fork@^2.2.1
MicroSD Card Adapter v1.0 11/01/2013 from CATLEX
I have a file with 4800KB on my SD card (32 SD HC Class10, but I tried others as well, they work in gopro and provide ~50mbps)
I only need to read these 4800bytes in less than about 5ms, which I think should work or even faster. But it takes 36ms that is kind of long in my opinion.

```
SdFat32 sd;
SpiPort_t spi2 = SpiPort_t(PB15, PB14, PB13);
SdSpiConfig config = SdSpiConfig(PB12, 0, SPI_FULL_SPEED, &spi2);
sd.begin(config);
digitalWrite(PIN_DT_READING, HIGH);  // for time measurement
File32 file = sd.open(DATA_FILE_PATH);
uint16_t *data[2400]; // I want 16bit/words, not bytes so I store it here
int data_read = file.readBytes((uint8_t *)data, 4800); // returns 4800 [bytes read]
digitalWrite(PIN_DT_READ, LOW); // for time measurement (pin is high for ~36ms)
```

sd.vol()->fatType() returns 32
When I look at the MISO2/MOSI2/SCK2/CS pins with a logic analyzer I see that the clock is not regularly beating but chunked.
[image]

And half of the time it is dilly dally doing nothing
[image]

I don't know what to do.

Moreover I found out something verry bizzare, when I instead call "file.readBytes((uint8_t *)data, 4801);"
(pretending that the target data buffer is bigger than the data in the file) then the whole process works as well and fills 4800 bytes as the file only has that much, but takes ~1050ms instead, which is super extremely strange in my opinion, maybe some problem with the filestream class, and maybe a missing flush()?

Do I have to provide more details?

ladyada, 2022-12-30:
we dont write any code/support for stm32 bluepill - you could try the original version
https://github.com/greiman/SdFat
but we have no reccomendation as we only made changes for Qspi flash addition

closed 2022-12-30 by ladyada

nablabla, 2022-12-30:
oh dang it. Thanks for the answer and suggestion! Just wanted to mention that the supported platforms lists it as supported, am I reading it wrong?